r-lidar/lasR #198: GDAL dataset creating error

ksemchh, 2025-08-27:
I've catch this error while processing data by pylasr:
```
  File "/app/TCD/pipeline.py", line 513, in _tree_detection
    pipeline.execute(las_files)
RuntimeError: error 6 while creating GDAL dataset. tree_crowns.tif: File too large regarding tile size. This would result in a file with tile arrays larger than 2GB
```
Therefore i've been see files that are processing in `las_files` variable:
```
(base) root@099c50f919d3:/tmp/tmpt6jabvsr/las# ls -lh
total 580M
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_1.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_10.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_11.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_12.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_13.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_14.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_15.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_16.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_17.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_18.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_19.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_2.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_20.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_21.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_22.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_23.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_24.laz
-rw-r--r-- 1 root root 155M Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_25.laz
-rw-r--r-- 1 root root 170M Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_26.laz
-rw-r--r-- 1 root root 161M Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_27.laz
-rw-r--r-- 1 root root  96M Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_28.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_3.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_4.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_5.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_6.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_7.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_8.laz
-rw-r--r-- 1 root root 1.9K Aug 27 07:55 tile_OR_UpperJohnDay_1_2020_9.laz
```

Total size of processing files is 580 Mb that are less than 2 Gb which are mentioned in error.

The pipeline that are processing:
```
-----------
other pipeline stages ...
-----------
pit_fill (uid:53551f83a8bd)
  thr_spk : -0.10 
  thr_lap : 0.10 
  med_size : 3 
  lap_size : 3 
  connect : 66aa2a101fb3 
  filter : [] 
  dil_radius : 0 
  output :  
-----------
local_maximum (uid:380f237ab072)
  record_attributes : false 
  min_height : 3.00 
  use_attribute : Z 
  ws : 4.00 
  filter : [] 
  output : /app/test1/chunk_0/tree_tops.gpkg 
-----------
region_growing (uid:aa8e107f876e)
  max_cr : 20.00 
  th_cr : 0.55 
  th_seed : 0.45 
  th_tree : 2.00 
  connect2 : 380f237ab072 
  connect1 : 53551f83a8bd 
  filter : [] 
  output : /app/test1/chunk_0/tree_crowns.tif 
-----------
```
Have you some ideas how resolve that error?
P.S. I've try to use old version of library before merging `api` branch into `main` and that error isn't appeared in same situation.

Jean-Romain, 2025-08-27:
I need a reproductible example.

Jean-Romain, 2025-08-27:
Ok I reproduced with 

```r
library(lasR)
f = "~/data/"
chm = rasterize(res=0.5)
pipeline = triangulation + chm
exec(pipeline, on = f, progress = T)
```

Your collection of files spans two continents, from North America to Africa. As a result, you are effectively requesting a raster that covers half the earth with a resolution of 50 cm

Looking at the files, I noticed that many of the headers are either corrupted, invalid, or not populated at all. Most of them report:

- 0 point records
- [0,0,0,0] bounding box (in Africa according to the CRS used)

These files are only ~1.9 KB in size, which means they are truly empty.

I did find 3 non-empty files, all located in Oregon, USA. If you remove the empty files, your code should work as expected.

As an improvement, I could update `lasR` so that it automatically discard zero-point files on the fly.

```r
library(lidR)
f = "~/data/"
ctg = readLAScatalog(f)
ctg
ctg = ctg[ctg[["Number.of.point.records"]] > 0,]
plot(ctg, map = T)
``

Jean-Romain, 2025-08-27:
0 point files are now discarded with a warning printed.

commit 1c4be1cf (2025-08-27): Fix #198