r-lidar/lasR #204: ERROR: 'end-of-file during chunk with index'

wiesehahn, 2025-09-01:
I am encountering an error like this or similar from time to time. However I cant really reproduce since the error appears, but if we directly rerun the piepline everything works as expected. 
```r
ERROR: ERROR: 'end-of-file during chunk with index 763' after 38150000 of 83833055 points for 'some_file.laz'
ERROR: ERROR: cannot open file 'some_file.laz' for read
ERROR: ERROR: cannot open lasreaderlas with file name 'some_file.laz'
ERROR: ERROR (LASindex): cannot open file 'some_file.lax' for write
Fehler: in 'write_lax' while processing the catalog: LASlib internal error
```

This time when running `exec(write_lax(embedded = TRUE), on = vpc)`, but I think I encountered it with a different pipeline the last time.

Jean-Romain, 2025-09-01:
I'm keeping it opened. But without reproducible example I can't do anything.

wiesehahn, 2025-09-17:
It seems to work with lidR in contrast
```r
> las_in <- readALS(in_file)
> out_file <- writeLAS(las_in, templas())
> readLASheader(out_file)@PHB$`Min Y`
[1] 5727000
```