r-lidar/lasR #205: write_vpc() seems not to recognize absolute_path argument

wiesehahn, 2025-09-02:
When writing a Virtual Point Cloud the absolute_path argument seems not to have an effect. This might be a Windows issue as the path seems always absolute if files are on a different drive and always relative otherwise.

``` r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")

show_filenames <- function(vpc){
  t <- yyjsonr::read_json_file(vpc)
  filenames <- sapply(t$features$assets, function(x) x$data$href)
  
  return(filenames)
}

library(lasR)
#> Warning: Paket 'lasR' wurde unter R Version 4.5.1 erstellt

vpc_rel <- exec(
  write_vpc(tempfile(fileext = ".vpc"), 
            absolute_path = FALSE), 
  on = LASfile)

vpc_abs <- exec(
  write_vpc(tempfile(fileext = ".vpc"), 
            absolute_path = TRUE), 
  on = LASfile)

show_filenames(vpc_abs)
#> [1] "./../../../../../jwiesehahn/AppData/Local/Programs/R/R-4.5.0/library/lidR/extdata/Megaplot.laz"
show_filenames(vpc_rel)
#> [1] "./../../../../../jwiesehahn/AppData/Local/Programs/R/R-4.5.0/library/lidR/extdata/Megaplot.laz"

TMPfile <- fs::path("Y:/Jens/tmp/Megaplot.laz")
fs::file_copy(LASfile, TMPfile)
#> Error: [EEXIST] Failed to copy 'C:/Users/jwiesehahn/AppData/Local/Programs/R/R-4.5.0/library/lidR/extdata/Megaplot.laz' to 'Y:/Jens/tmp/Megaplot.laz': file already exists

vpc_rel <- exec(
  write_vpc(tempfile(fileext = ".vpc"), 
            absolute_path = FALSE), 
  on = TMPfile
)

vpc_abs <- exec(
  write_vpc(tempfile(fileext = ".vpc"), 
            absolute_path = TRUE), 
  on = TMPfile
)

show_filenames(vpc_abs)
#> [1] "Y:/Jens/tmp/Megaplot.laz"
show_filenames(vpc_rel)
#> [1] "Y:/Jens/tmp/Megaplot.laz"
```

<sup>Created on 2025-09-02 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-09-02:
You  are writing a VPC file in `C:`using `tempfile` with files in `Y:`. There is no relative path between different drive letters on Windows. The behavior of the filesystem is correct. Unlike linux or macOS windows can have different roots and you cannot "link" across drives

wiesehahn, 2025-09-02:
didnt knew that

Jean-Romain, 2025-09-02:
Ho I see. I missed this one.

Jean-Romain, 2025-09-02:
Fixed. Typo in the code introduced in v0.17.0

commit 1047663f (2025-09-02): Fix #205
closed 2025-09-02 by Jean-Romain