r-lidar/lasR #215: file is read/written only in parts

wiesehahn, 2025-09-17:
As mentioned in https://github.com/r-lidar/lasR/issues/204 I have a file which seems to be read/written only in parts by lasR.

```r
> library(lasR)
> in_file <- file_classified
> out_file <- exec(write_las(), on = in_file)
>
> info(in_file)
Source       : LASF (v1.4)
Size         : 3.45 GB
Extent       : 549000.00 550000.00 5727000.00 5728000.00 (xmin, xmax, ymin, ymax)
Points       : 53.98 millions
Area         : 999998.0 m²
Density      : 54.0 pts/m²
Coord. ref.  : ETRS89 / UTM zone 32N
Schema       :
31 attributes | 64 bytes per points
 Name: flags             | uchar  | Desc: Internal 8-bit mask reserved for lasR core engine
 Name: X                 | int    | Desc: X coordinate
 Name: Y                 | int    | Desc: Y coordinate
 Name: Z                 | int    | Desc: Z coordinate
 Name: Intensity         | ushort | Desc: Pulse return magnitude
 Name: ReturnNumber      | uchar  | Desc: Pulse return number for a given output pulse
 Name: NumberOfReturns   | uchar  | Desc: Total number of returns for a given pulse
 Name: Classification    | uchar  | Desc: The 'class' attributes of a point
 Name: UserData          | uchar  | Desc: Used at the user’s discretion
 Name: PointSourceID     | short  | Desc: Source from which this point originated
 Name: ScanAngle         | float  | Desc: Angle at which the laser point was output
 Name: ScannerChannel    | uchar  | Desc: Channel (scanner head) of a multi-channel system
 Name: gpstime           | double | Desc: Time tag value at which the point was observed
 Name: R                 | ushort | Desc: Red image channel
 Name: G                 | ushort | Desc: Green image channel
 Name: B                 | ushort | Desc: Blue image channel
 Name: NIR               | ushort | Desc: Near infrared channel value
 Name: Amplitude         | ushort | Desc: Echo signal amplitude [dB]
 Name: Pulse width       | ushort | Desc: Full width at half maximum [ns]
 Name: Reflectance       | short  | Desc: Echo signal reflectance [dB]
 Name: Deviation         | ushort | Desc: Pulse shape deviation
 Name: confidence        | uchar  | Desc: confidence values
 Name: Distance          | int    | Desc: Distance
 Name: Group             | uint   | Desc: Group
 Name: Normal            | uint   | Desc: Normal vector 2+15+15 bits
 Name: EdgeOfFlightline  | bit    | Desc: Set when the point is at the end of a scan
 Name: ScanDirectionFlag | bit    | Desc: Direction in which the scanner mirror was traveling 
 Name: Synthetic         | bit    | Desc: Point created by a technique other than direct observation
 Name: Keypoint          | bit    | Desc: Point is considered to be a model key-point
 Name: Withheld          | bit    | Desc: Point is supposed to be deleted)
 Name: Overlap           | bit    | Desc: If set, point is within an overlap region of 2+ swaths
>
> info(out_file)
Source       : LASF (v1.4)
Size         : 3.08 GB
Extent       : 549000.00 550000.00 5727100.00 5728000.00 (xmin, xmax, ymin, ymax)
Points       : 48.15 millions
Area         : 899997.1 m²
Density      : 53.5 pts/m²
Coord. ref.  : ETRS89 / UTM zone 32N
Schema       :
31 attributes | 64 bytes per points
 Name: flags             | uchar  | Desc: Internal 8-bit mask reserved for lasR core engine
 Name: X                 | int    | Desc: X coordinate
 Name: Y                 | int    | Desc: Y coordinate
 Name: Z                 | int    | Desc: Z coordinate
 Name: Intensity         | ushort | Desc: Pulse return magnitude
 Name: ReturnNumber      | uchar  | Desc: Pulse return number for a given output pulse
 Name: NumberOfReturns   | uchar  | Desc: Total number of returns for a given pulse
 Name: Classification    | uchar  | Desc: The 'class' attributes of a point
 Name: UserData          | uchar  | Desc: Used at the user’s discretion
 Name: PointSourceID     | short  | Desc: Source from which this point originated
 Name: ScanAngle         | float  | Desc: Angle at which the laser point was output
 Name: ScannerChannel    | uchar  | Desc: Channel (scanner head) of a multi-channel system
 Name: gpstime           | double | Desc: Time tag value at which the point was observed
 Name: R                 | ushort | Desc: Red image channel
 Name: G                 | ushort | Desc: Green image channel
 Name: B                 | ushort | Desc: Blue image channel
 Name: NIR               | ushort | Desc: Near infrared channel value
 Name: Amplitude         | ushort | Desc: Echo signal amplitude [dB]
 Name: Pulse width       | ushort | Desc: Full width at half maximum [ns]
 Name: Reflectance       | short  | Desc: Echo signal reflectance [dB]
 Name: Deviation         | ushort | Desc: Pulse shape deviation
 Name: confidence        | uchar  | Desc: confidence values
 Name: Distance          | int    | Desc: Distance
 Name: Group             | uint   | Desc: Group
 Name: Normal            | uint   | Desc: Normal vector 2+15+15 bits
 Name: EdgeOfFlightline  | bit    | Desc: Set when the point is at the end of a scan
 Name: ScanDirectionFlag | bit    | Desc: Direction in which the scanner mirror was traveling 
 Name: Synthetic         | bit    | Desc: Point created by a technique other than direct observation
 Name: Keypoint          | bit    | Desc: Point is considered to be a model key-point
 Name: Withheld          | bit    | Desc: Point is supposed to be deleted)
 Name: Overlap           | bit    | Desc: If set, point is within an overlap region of 2+ swaths
```

the file can be downloaded at https://e.pcloud.link/publink/show?code=XZ6JrOZ29e5BghFyKhgvSTT0Yqp4HMmcBkX

Jean-Romain, 2025-09-17:
I get 53.98 millions points both input output

wiesehahn, 2025-09-17:
this is extremely weird, it seems I can reproduce sometimes and then not and I am unsure why. I downloaded the file as well and it worked. Now one last try, here is a file with which I just reproduced this three times in a row, although I am quite sure it should be the same file as before and I am also quite sure that it worked without problem once. 

https://e.pcloud.link/publink/show?code=XZ6JrOZ29e5BghFyKhgvSTT0Yqp4HMmcBkX

Jean-Romain, 2025-09-17:
Random behavior are usually a hint for memory access corruption. The problem is that I cannot try to search memory corruption with a 3GB point cloud. The tools for debugging such trouble are slow. In need to reproduce with smaller files.

wiesehahn, 2025-09-19:
just for my understanding, this is a problem/bug with the data or the writing software (lastools in this case) and not the reading software (lasR)?

I just tried to run the pipeline again, and in one case it seemed to work. As I just had the problem with a single file I just keep the random proper result and am fine for the moment.

Jean-Romain, 2025-09-19:
Is it a question or an affirmation? Did you write the file with LAStools? Your example is a write with lasR.

wiesehahn, 2025-09-19:
The file was classified (ground) with lastools in a previous step. Then we have trouble sometimes, lasR reads/writes only parts of it in some cases. Actually I have not testet if we have this behaviour with the unclassified file as well or if if we apply some read/write processes with lastools on the classified file. I might do this later, currently I am fine with the result and have not much time to fiddle further, so this was just a question for my understanding. If this is just data curruption which does not have to do anything with lasR we can close this.

Jean-Romain, 2025-11-05:
Closing. Could be reopened with a reproducible exemple